arkypita/LaserGRBL #2241: I have no idea what's happening

Giosca0502, 2024-04-16:
When I try to pair my engraver it gives me this back

16/04/2024 14:32:31.123	SetStatus   	Machine status [Connecting]
16/04/2024 14:32:31.129	OpenCom     	Open COM11 @ 9600 baud Ctrl-X (UsbSerial)
16/04/2024 14:32:33.675	SendImmediate	Send Immediate Command [0x18]
16/04/2024 14:32:41.130	OpenCom     	Connection timeout!
16/04/2024 14:32:41.130	CloseCom    	Close COM11 [CORE]
16/04/2024 14:32:41.134	SetStatus   	Machine status [Disconnected]

And Idk what to do
Yes, I have updated both LaserGRBL and CH340

arkypita, 2024-04-16:
The machine is not responding, LaserGRBL close the connection due to timeout (10s)

Are you sure you are using the right baudrate?

I see 9600, but most machine use 115200

Giosca0502, 2024-04-16:
Well, the machine is an Atomstack A5 Pro, that works with 115200 baudrate, but I'm trying to connect a CH-06 wired to the main board, which has a 9600 baudrate