arkypita/LaserGRBL #2251: desconectando assim que inicia o corte

3105ferr, 2024-05-16:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware and software configuration (please complete the following information):**
 - Windows version
 - Engraver brand and model
 - Grbl version (i.e. 0.9j, 1.1f)
 - LaserGRBL version

**Additional context**
Add any other context about the problem here.

3105ferr, 2024-05-16:
esta desconectando assim q inicia o corte

ristraus, 2024-05-20:
Por favor de mais detalhes, conforme solicitado acima.