arkypita/LaserGRBL #2254: Exception in 6.9.9

melvinisken, 2024-05-22:
I got the following error message right after installing 6.9.9 and starting it directly from the installer (it tells to upload here, so I do): 
[image]

```
LaserGrbl v6.9.9.0
Grbl v1.1f
Wrapper: UsbSerial
Win 11, Professional, 10.0.22631.0 (7)
CLR: C:\Windows\Microsoft.NET\Framework64\v4.0.30319\mscorlib.dll 4.8.9241.0 built by: NET481REL1LAST_C

TypeOf exception  [System.NullReferenceException]
Exception message [Object reference not set to an instance of an object.]
Exception source  [LaserGRBL], thread [Main Thread]
Exception method  [Void OnPaint(System.Windows.Forms.PaintEventArgs)]

   ----------- stack trace -----------
   at LaserGRBL.UserControls.ImageButton.OnPaint(PaintEventArgs e) in E:\LaserGRBL\LaserGRBL\UserControls\ImageButton.cs:line 215
   at LaserGRBL.PreviewForm.CustomButtonIB.OnPaint(PaintEventArgs e) in E:\LaserGRBL\LaserGRBL\PreviewForm.cs:line 255
   at System.Windows.Forms.Control.PaintWithErrorHandling(PaintEventArgs e, Int16 layer)
   at System.Windows.Forms.Control.WmPaint(Message& m)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.UserControl.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

```

osbrook, 2024-05-22:
Me too but Windows 10
LaserGrbl v6.9.9.0
Grbl v
Wrapper: UsbSerial
Win 10, Home, 10.0.19045.0 (7)
CLR: C:\Windows\Microsoft.NET\Framework64\v4.0.30319\mscorlib.dll 4.8.9232.0 built by: NET481REL1LAST_C

TypeOf exception  [System.NullReferenceException]
Exception message [Object reference not set to an instance of an object.]
Exception source  [LaserGRBL], thread [Main Thread]
Exception method  [Void OnPaint(System.Windows.Forms.PaintEventArgs)]

   ----------- stack trace -----------
   at LaserGRBL.UserControls.ImageButton.OnPaint(PaintEventArgs e) in E:\LaserGRBL\LaserGRBL\UserControls\ImageButton.cs:line 215
   at LaserGRBL.PreviewForm.CustomButtonIB.OnPaint(PaintEventArgs e) in E:\LaserGRBL\LaserGRBL\PreviewForm.cs:line 255
   at System.Windows.Forms.Control.PaintWithErrorHandling(PaintEventArgs e, Int16 layer)
   at System.Windows.Forms.Control.WmPaint(Message& m)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.UserControl.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

arkypita, 2024-05-23:
Please update to v6.9.10 and let me know: https://github.com/arkypita/LaserGRBL/releases/download/v6.9.10/install.exe

melvinisken, 2024-05-23:
> Please update to v6.9.10 and let me know: https://github.com/arkypita/LaserGRBL/releases/download/v6.9.10/install.exe

Yes, that seems to have solved it. No error message. 

Another small remark (does not fit in this issue, but as I'm writing, I thought I mention it. **In dark modes, the bottom left sliders do not look deactivated** but active. Though, you can't use them. I suggest changing the color to dark gray like the move arrows :-) 
[image]