arkypita/LaserGRBL #2276: About 10% cpu usage on idle

Whateverman2012, 2024-06-14:
Hi,
  I'm not sure if it is a bug but I get about 10% cpu usage even at start, no file, no laser device connected.
  I notice it cause my laptop's fan kicks in when it is over 8 % cpu usage and was wondering why while I was not using it.

 - Windows version : 11
 - Engraver brand and model : Sculpfun S30 pro
 - Grbl version : 1.1f
 - LaserGRBL version 7.6.0
 -  i7-8750H CPU @ 2.20GHz, 16gb ram, GTX 1070

  Regards

StuartB4, 2024-06-14:
That seems high 10%. Just checked mine using windows 10 and even with 3 other apps running and my internet browser it does'nt go over 4% when it's running without anything connected. 
Do you have any other apps running in the background that don't need to be?

arkypita, 2024-06-14:
Good morning

The old version 6.x drew the preview only once, at startup or when loading the image. Then stop.

The new 7.x version continuously redraws the job preview (as well as number grids and location data) using a worker thread. This is to allow the interaction of zoom and pan, and real-time coloring of the work done.

This obviously requires greater use of PC resources which, where there is a modern multi-core processor, can be truly negligible (4% - 6%). However, where there are slightly older processors, CPU consumption as a percentage of total resources is higher (10 - 20%).

In any case this drawing thread "releases" resources to avoid saturating the processor, and implements adaptive techniques to balance performance.

You can feel comfortable with this 10%, however we are working to insert further optimizations and reduce CPU time where redrawing is superfluous.

Best regards

Whateverman2012, 2024-06-15:
Hi,

  First, thank you for that pretty good piece of software.
  I use it along Rhino and python to make laser cutting a lot more enjoyable compared to Lightburn.
  A simple keymapped reload file function made my working pipeline easy.

  It seems really well made.
  Congrats to you and all contributors !

  After further investigation, it is more 5% than 10%.
  I used Process Explorer to check computer usage.
  I will clearly follow your progression and take a look at version 6.

  I'll take this opportunity to make a tiny suggestion.
  A custom keymapping to trigger laser target position instead of the double-click would be nice.
  I agree it is subtile but double-clicking is kind of unprecise due to the possible drift of the mouse.
  Personally I've added a control key modifier to a click event in a local fork of 6.9.12.
 
  Thank for your time and your thorough explanations.

  Regards

closed 2024-06-15 by Whateverman2012