arkypita/LaserGRBL #2281: gcode header is not in files

kleffa, 2024-06-17:
**Describe the bug**
the configured header is not present when file is saves

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Settings, GCODE, Header
2. set header for $H (I used G28, but newest vertion just give me trobles with hard limits)
3. Go file, save program
4. open ouput file

**Expected behavior**
ouput file must have that header

**Hardware and software configuration (please complete the following information):**
 - Windows 11
 - LaserGRBL 7.7

**Additional context**
this worked in older versions

arkypita, 2024-06-18:
This is by design. If you save gcode with header and footer (that are automatically added/sent when running a job) the results is that when you load it back you send it twice.

However look at the menu, there is an advanced save options where you can set to export header and footer too.

kleffa, 2024-06-18:
👍🏽

closed 2024-06-18 by kleffa