arkypita/LaserGRBL #2296: Error

Micaela002, 2024-06-26:
LaserGrbl v7.8.0.0
Grbl v1.1f
Wrapper: UsbSerial
Win 7, Ultimate, 6.1.7601.65536, Service Pack 1 (7)
CLR: C:\Windows\Microsoft.NET\Framework64\v4.0.30319\mscorlib.dll 4.8.4110.0 built by: NET48REL1LAST_B

TypeOf exception  [System.OutOfMemoryException]
Exception message [Memoria insuficiente.]
Exception source  [System.Drawing], thread [Main Thread]
Exception method  [Void CheckErrorStatus(Int32)]

   ----------- stack trace -----------
   en System.Drawing.Graphics.CheckErrorStatus(Int32 status)
   en System.Drawing.Graphics.DrawImage(Image image, Rectangle destRect, Int32 srcX, Int32 srcY, Int32 srcWidth, Int32 srcHeight, GraphicsUnit srcUnit, ImageAttributes imageAttrs, DrawImageAbort callback, IntPtr callbackData)
   en System.Drawing.Graphics.DrawImage(Image image, Rectangle destRect, Int32 srcX, Int32 srcY, Int32 srcWidth, Int32 srcHeight, GraphicsUnit srcUnit, ImageAttributes imageAttr)
   en System.Windows.Forms.ControlPaint.DrawBackgroundImage(Graphics g, Image backgroundImage, Color backColor, ImageLayout backgroundImageLayout, Rectangle bounds, Rectangle clipRect, Point scrollOffset, RightToLeft rightToLeft)
   en System.Windows.Forms.Control.PaintBackground(PaintEventArgs e, Rectangle rectangle, Color backColor, Point scrollOffset)
   en System.Windows.Forms.Control.PaintBackground(PaintEventArgs e, Rectangle rectangle)
   en System.Windows.Forms.Control.OnPaintBackground(PaintEventArgs pevent)
   en System.Windows.Forms.ScrollableControl.OnPaintBackground(PaintEventArgs e)
   en System.Windows.Forms.Control.PaintWithErrorHandling(PaintEventArgs e, Int16 layer)
   en System.Windows.Forms.Control.WmPaint(Message& m)
   en System.Windows.Forms.Control.WndProc(Message& m)
   en System.Windows.Forms.UserControl.WndProc(Message& m)
   en System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

arkypita, 2024-06-27:
Hi 
Latest versions of LaserGRBL have added 3D graphics which can be very memory hungry especially with very line-heavy graphics work, and this can be a problem on older PCs.

There is the possibility to run LaserGRBL in a lighter "compatible" mode permanently simply by configuring "Legacy" in the LaserGRBL options, or temporarily by launching the program with the "LaserGRBL (disable opengl)" link from windows menu.

[image]

CrasherJK, 2024-07-12:
LaserGrbl v7.11.0.0
Grbl v1.1h
Wrapper: UsbSerial
Win 11, Professional, 10.0.22000.0 (7)
CLR: C:\Windows\Microsoft.NET\Framework64\v4.0.30319\mscorlib.dll 4.8.9195.0 built by: NET481REL1LAST_B

TypeOf exception  [System.ObjectDisposedException]
Exception message [Bezpečný popisovač byl zavřen.]
Exception source  [mscorlib], thread [Main Thread]
Exception method  [Void DangerousAddRef(Boolean ByRef)]

   ----------- stack trace -----------
   v System.Runtime.InteropServices.SafeHandle.DangerousAddRef(Boolean& success)
   v System.StubHelpers.StubHelpers.SafeHandleAddRef(SafeHandle pHandle, Boolean& success)
   v Microsoft.Win32.UnsafeNativeMethods.GetOverlappedResult(SafeFileHandle hFile, NativeOverlapped* lpOverlapped, Int32& lpNumberOfBytesTransferred, Boolean bWait)
   v System.IO.Ports.SerialStream.EventLoopRunner.WaitForCommEvent()
   v System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   v System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   v System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
   v System.Threading.ThreadHelper.ThreadStart()