arkypita/LaserGRBL #2297: v7.8.0 does not load raster image

misterunix, 2024-06-26:
**Describe the bug**
When loading a png file, the raster import dialog comes up, but there is a bouncing bar in the lower right corner. Can not cancel, close nor chose next. The only option is to terminate the program. Env: Linux Ubuntu 22.04 running on playonlinux.

**To Reproduce**
Steps to reproduce the behavior:
1. File open
2. pick a png image
3. open

**Expected behavior**
I would expect the dialog to move on to the next step.

**Screenshots**
[image]
notice lower right corner - that blew square never stops bouncing back n forth,

**Hardware and software configuration (please complete the following information):**
 - Linux Big-Dell 6.5.0-41-generic #41~22.04.2-Ubuntu SMP PREEMPT_DYNAMIC Mon Jun  3 11:32:55 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
 - Longer Ray5 20W - currently not connected.
 - ?
 - ?

**Additional context**
I am REALLY new to this. The only windows machine in the house is my wife's, so I need to make this working undex linux.

arkypita, 2024-06-27:
It is nice to see that you are using LaserGRBL on Linux machine.

LaserGRBL was developed to be a Windows software, so you can expect some issue on linux, howhever if it is possible i like to have it run on Linux too.

Can you explain me how you have installed/configured your linux machine to be able to run LaserGRBL, so i can make some test too?

misterunix, 2024-06-27:
I installed playonliux. Then "+" to install a program. 
Lower left of the dialog box is "Install a non-listed program"
Bunch of info dump boxes, next,next,next
In dialog window "Install a program in a new virtual drive" , next
Give the "virtual drive a name" Mine is LaserGRBL2"
I am then 
- Use another version of wine
- Configure Wine
Next
Choosing Wine 5.22
Next
64 bits windows installation
In my case it downloads wine 5.22 again, takes time to install and configure. 
Configure wine dialog, verify drives, then "OK"
Back to playonlinux
"Please select the install file to run" - Browse
Navigate to the install.exe file and select it.
Next
Go through the diaglogs, do not launch when complete.
Select shortcut name "LaserGRBL.exe"
Next
Next
"I don't want to make another shortcut"
Next

Now run it

File Open
Select a png file
the blue square will sit there and bounce

Hope that made sense.

misterunix, 2024-06-27:
I tried a version 6 install. Same problem.

arkypita, 2024-06-27:
If you go to menu ?, open session log, did you see any error?

misterunix, 2024-06-27:
Once the blue box starts bouncing, you cant get to the ?.

This what I am able to get.
[lasergrbl-1.txt](https://github.com/user-attachments/files/16021573/lasergrbl-1.txt)

misterunix, 2024-06-28:
I got it to load a PNG file. I made many config changes, so I am still determining which setting or settings did the trick. I will install a new version and document everything I do so that, hopefully, it will help you. It's Friday, so give me a day to get it together. 
Better yet, if there is something I can do that will dump all the config settings to a file...

misterunix, 2024-06-28:
Works
- 1 bit BW Dithering 
- Passthrough

Fails
- Vectorize!
- Cant check Centerline, even built as a 64b wine install, still detects as 32b.

Now I need to put the laser cutter back on the table and give it a go.

misterunix, 2024-07-01:
Closed as user error/.net on linux.

closed 2024-07-01 by misterunix

charely6, 2024-09-13:
Hey I ran into the same issue
is my logs from the session.log.

12/09/2024 20:51:21.004	OpenGL      	AUTO->FBO OpenGL 4.2 (Compatibility Profile) Mesa 24.0.9-0ubuntu0.1, Intel, Mesa Intel(R) HD Graphics 4000 (IVB GT2)
12/09/2024 20:51:41.031	OpenFile    	Open C:\users\charely6\Documents\KeychainTestBorders.svg
12/09/2024 20:52:33.883	OpenFile    	Open C:\users\charely6\Documents\KeychainTestBorder.svg
12/09/2024 20:52:44.443	OpenFile    	Open C:\users\charely6\Documents\KeychainTest.png
12/09/2024 20:54:59.666	Program     	---------------- PROGRAM STOP -----------------

12/09/2024 21:04:26.279	Program     	------- LaserGRBL v7.12.0 [64bit] START -------
12/09/2024 21:04:27.591	OpenGL      	AUTO->FBO OpenGL 4.2 (Compatibility Profile) Mesa 24.0.9-0ubuntu0.1, Intel, Mesa Intel(R) HD Graphics 4000 (IVB GT2)

The "PROGRAM STOP" is from when I killed it using task manager before I realized that the other settings let something work but it didn't note anything else? 

when its set to "Line to Line tracing" the little loading bar over the next button just pings back and forth and the next button doesn't work.
Also the preview image doesn't show up until I start changing settings just shows a blank checkerboard.

If you need anything else please let me know, I'm planning to use this on linux so I can use an older laptop as my dedicated laser engraver machine and its pretty old at this point.

Akrobate, 2024-10-22:
Hello,

I have the same exactly same behaviour.

I'm using LaserGRBL v7.12.0 on Ubuntu 22.04 with wine

There is nothing interresting in sessionlog.

I used this ressource to help me to install it on ubuntu: https://www.aranacorp.com/fr/installer-lasergrbl-sous-linux/

As others I can get the image rendered with 1bit

[image]

And the line by line doesn't load

[image]

Do you need some more information to debug?

charely6, 2024-10-22:
I found a different guide that included configuring the the usb device
stuff and after doing all that the raster thing seemed to work. I can look
for which guide that was later tonight