arkypita/LaserGRBL #2303: Missing Translations

ristraus, 2024-07-05:
**Describe the bug**
There are a few missing translations

**To Reproduce**
Just open the related menu option

LaserGRBL V 7.11

**Additional context**
1) No matter the language, ? - License is only shown in English, although LicenseForm.resx has translations.

2) Power vs Speed test generator is only available in English. \Generator\PowerVsSpeedForm.resx is empty.

3) Cuting Test generator is only available in English. \Generator\CuttingTest.resx is empty.

4) Laser & Material Database has only English colum headers. 

5) Laser Lifespan Monitoring Tool is only available in English. LaserLifeEdit.resx has translation but not loaded, same for LaserUsage.resx.

6) Precision Test color filter only shows colors in English.

7) Shake test is only available in English. although ShakeTest.resx has translations.

8) Preview Show Bounding Box, Show cross cursor and Auto size mode not translated (I just submited MainForm.pt-BR.resx)

9) Buttons descriptions has no translation.

arkypita, 2024-07-08:
Thanks for your detailed report!

When I add new interfaces it takes a little additional manual work to include them in the translation mechanism. Now that you've pointed it out, I'll make these forms translatable too. Best regards

arkypita, 2024-07-08:
> License is only shown in English

This is a choice. I prefer that the license is not translated for legal reasons.

ristraus, 2024-07-09:
I found an extra translation error:
[image]
it happens only in PT_BR
[image]
The translation file seems ok, and was loading correctly on older versions.
[image]
[image]

Wondering if I need to reduce to "Qualidade" only insted of "Qualidade do preenchimento" which is probably too large ?

Grazie

ristraus, 2024-07-24:
1 - OK
2 - Fixed on V 7.12.0
3 - Fixed on V 7.12.0
4 - Fixed on V 7.12.0
5 - Fixed on V 7.12.0
6 - Still Pending
7 - Fixed on V 7.12.0
8 - Fixed on V 7.12.0
9 - Still Pending

"Quality" issue solved with short translation text.

ristraus, 2024-07-24:
will create a new issue with the 2 pending itens plus some news.

closed 2024-07-24 by ristraus