arkypita/LaserGRBL #2332: No more network drive access

talik31, 2024-08-05:
Since version 7 the application is not capable anymore to load or access to files on a Windows network drive

**To Reproduce**
Create a network drive to a SMB sharing, then try to access to its content through file navigator in the application 
-> the drive isn't shown
Try to drag & drop a file from windows explorer.
-> the cursor is changed to a striked circle

**Expected behavior**
The network drives should be shown in the application file explorer and drag&drop of files from a network drive with windows explorer should be opened directly. (behavior of version before v7)

**Hardware and software configuration (please complete the following information):**
 - Windows 10 professional
 - LaserGRBL version : 7.12.0

**Out ot context**
Thank you so much to all the devs for this incredible application, you're the best 👍👍👍

arkypita, 2024-08-05:
Nothing has changed in how LaserGRBL handle file open, however file open is a form of windows OS, so not handled directly from LaserGRBL. Maybe a windows update has broken something?

Could you please make a cross test, installing old v6.2 and see if any difference? Then go back to 7.2 and test again. Thank you

talik31, 2024-08-13:
Hello, I've installed 6.2.2 on a w11 PC and performed a drag&drop of a svg file with success, then updated to 7.12.0 and the drag&drop isn't working anymore (even for local files in fact, it's not only network drive files). Please find attached videos of the 2 different behaviors.

https://github.com/user-attachments/assets/851727b7-930e-4642-9296-95ddb3d2370b

https://github.com/user-attachments/assets/b183fdd2-15b6-4247-82d5-6bb081fb59cd

arkypita, 2024-08-13:
There is a known windows bug. At the first install drag drop doesn't work. Close LaserGRBL and restart, after the first run, the drag drop work again. Let me know

However I'll double check it when I am back from vacation

talik31, 2024-08-13:
You're completely right, after closing & reopening the application the drag&drop feature is working again. Maybe I haven't restart the application after an upgrade in the other computer, but anyway if it's not working on my other computer it's obviously the OS which is causing the issue. The issue can be close.
Thank you so much once again for the wonderful work you're doing with this application 👍👍👍