arkypita/LaserGRBL #2339: CPU and DISK usage - LaserGRBL freezes - Not responding every few seconds

bwoostdam, 2024-08-15:
I am seeing high cpu usage and much to high constant disk usage.
I have read that the new constant 'redrawing' is responsible for this.

Ik think this makes LaserGRBL freeze every few seconds for several minutes and also does not send data to the machine.
After a few minutes of processing a few lines of Gcode is being send again to the machine.

The result is that the TX buffer seems to have a problem. I do not believe this to be true.
I am a software tester for a software company and I think this behaviour is not related to a comm-port 'buffer-overrun'.

I use an older machine with limited logical cores (i5) and limited RAM 8GB.
I try to work with larger GCode files (~20Mb) on a new NEJE 4 MAX - E80 with extension kit

To me it is very clear that the comm-port buffer is not overrun at all but LaserGRBL is too busy swapping memory to disk and vice versa.

Reproduce:  When the files fit within RAM there is no problem.  When the file-size increases and it starts to swap to disk the problem arises. The larger the file the worse the situation becomes.

I had to make it 01:45 last night to see my job finish... Started at 19:30  and should have taken no more than 1,5 hours.

How can I turn of this constant redrawing, it takes far too many resources.

arkypita, 2024-08-15:
Hi
I have tested it into an old XP laptop, very very poor in term of resources, with great results. If you are using latest version 7.12 redraw is done only if needed and using display-list optimization, so only a small amount of poligon (lines) are redrawn to update the executed job (lines that turn green).

I truly believe that going out of ram and swapping could be a big problem, and yes.. if you have done a cross test with old/new version and you can repeat your results that is no doubt that the new opengl (that in your pc could be a software implementation) is the cause of slowness.

P.S. the new version comes with compatibility options that allow to turn off opengl rendering and use old preview window. Did you have tried it?

Start LaserGRBL from windows menu, you find a new link. It is also possible to turn off permanently from LaserGRBL settings.

@GabeMx5

bwoostdam, 2024-08-15:
Thank you very much for your information. I will try the OpenGL option.

bwoostdam, 2024-08-15:
I did not know this option is available.  It loads the file very fast and
CPU and DISK usage is now normal!!
Oh I really hope this will be the solution.  I can try this tonight.  Thank
you!!

bwoostdam, 2024-08-15:
I was able to try your suggestion and it indeed works! I can now cut in one
continuous flow. Thank you.
I would like to advise that the setup should make 2 shortcuts to the
desktop. It will make it more obvious the option exists.
In Windows11, it is not very evident from the poor windows menu that there
is such an option.
Thanks again