arkypita/LaserGRBL #2348: Oooops! Something Went Wrong

YGSteve, 2024-08-26:
I am just following instructions after I finished one engraving. The following is what the program want me to past here:

LaserGrbl v7.12.0.0
Grbl v
Wrapper: UsbSerial
Win 11, Home, 10.0.22631.0 (7)
CLR: C:\Windows\Microsoft.NET\Framework64\v4.0.30319\mscorlib.dll 4.8.9256.0 built by: NET481REL1LAST_B

TypeOf exception  [System.ObjectDisposedException]
Exception message [Safe handle has been closed]
Exception source  [mscorlib], thread [Main Thread]
Exception method  [Void DangerousAddRef(Boolean ByRef)]

   ----------- stack trace -----------
   at System.Runtime.InteropServices.SafeHandle.DangerousAddRef(Boolean& success)
   at System.StubHelpers.StubHelpers.SafeHandleAddRef(SafeHandle pHandle, Boolean& success)
   at Microsoft.Win32.UnsafeNativeMethods.GetOverlappedResult(SafeFileHandle hFile, NativeOverlapped* lpOverlapped, Int32& lpNumberOfBytesTransferred, Boolean bWait)
   at System.IO.Ports.SerialStream.EventLoopRunner.WaitForCommEvent()
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Threading.ThreadHelper.ThreadStart()

Ellias68, 2024-09-10:
LaserGrbl v7.12.0.0
Grbl v1.1h
Wrapper: UsbSerial
Win 10, Home, 10.0.19045.0 (7)
CLR: C:\Windows\Microsoft.NET\Framework64\v4.0.30319\mscorlib.dll 4.8.9261.0 built by: NET481REL1LAST_C

TypeOf exception  [System.ObjectDisposedException]
Exception message [Safe handle has been closed]
Exception source  [mscorlib], thread [Main Thread]
Exception method  [Void DangerousAddRef(Boolean ByRef)]

   ----------- stack trace -----------
   at System.Runtime.InteropServices.SafeHandle.DangerousAddRef(Boolean& success)
   at System.StubHelpers.StubHelpers.SafeHandleAddRef(SafeHandle pHandle, Boolean& success)
   at Microsoft.Win32.UnsafeNativeMethods.GetOverlappedResult(SafeFileHandle hFile, NativeOverlapped* lpOverlapped, Int32& lpNumberOfBytesTransferred, Boolean bWait)
   at System.IO.Ports.SerialStream.EventLoopRunner.WaitForCommEvent()
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Threading.ThreadHelper.ThreadStart()

Ellias68, 2024-09-10:
Also, i am facing few issues, x and y are switched, the shape are distorted; a circle is  oval... distances are not proportional...I guess this is a typical to newbie