arkypita/LaserGRBL #2352: preview image cropping seems to not crop to crop line exactly

goofyseeker311, 2024-09-01:
**Describe the bug**
cropping on the right bottom corner is way off, or the preview borders hide part of the image.

**To Reproduce**
Steps to reproduce the behavior:
1. open and crop a very small part of preview image
2.  dont see what you wanted in the preview crop window

**Expected behavior**
expected view is to see everything pixel accurate what you selected, exact cropping borders.

**Screenshots**
the crop was meant to be exactly symmetric on both sides, but more was cut from right and bottom.
[image]

**Hardware and software configuration (please complete the following information):**
 - win11
 - v7.12.0

**Additional context**
image tools should make exact edits to the preview window or at least show them like they are.

goofyseeker311, 2024-09-02:
some other issues: when pressing quick save, the file name is prefixed with "C:\Users\[username]\AppData\Local\Temp\" even if your current directory in the dialog is somewhere else than the user temp directory.

Thebear66, 2025-01-05:
To me, when cropping, the frame that should help to cut the picture goes completely outside of it and I cannot only crop blindly. And actually, the common use of cropping is moving handles to the center of the picture, not to draw a frame. 

[image]