arkypita/LaserGRBL #2373: 5 Seconds Countdown with anoying beep sound

HeldHannes, 2024-09-28:
Hi,

Intro:
My computer crashed (not because of LASERGRBL) and I had to get a new one,
so I thought - why not update LASERGRBL to a newer version (~6.2.???).

Plot:
I understand the idea of trying to be as safe as one can get, but I do not want to be treated like a child.
I know that a laser is a dangerous weapon of mass destruction, but I don't like to reminded about it before every operation.
I tried to shut of every alarm and sound of LASERGRBL, but it still beeps when I press START.

Solution-request:
I'd very like to be able to disable the 5-sec-warning and especially the annoying sound that it is producing.

Thank you very much.

Hostaler, 2024-09-28:
In options 

[image]

HeldHannes, 2024-09-28:
Thanks,

my options-tab looks like:
[image]

HeldHannes, 2024-09-28:
Is this a language-setting-issue? 
or is my version already outdated?

HeldHannes, 2024-09-28:
Sound-tab looks OK:

[image]

HeldHannes, 2024-09-28:
Dark-Design-Problem

I can check and see the text only on light design.

[image]

closed 2024-09-28 by HeldHannes