arkypita/LaserGRBL #2390: PostScript after Success or Error

AlexAwesome88, 2024-10-22:
@arkypita 

I'd like an option to execute a postscript after finishing. Maybe with passing a parameter for success or error. For example running a batch script or just let us choose a script or an executable that shall be run.

My problem is: I'd like to send notifications to my phone via REST API which is super easy for example with python or simply using curl because there are a lot of Apps that you can send notifications to via REST.

I'd be super happy and I'm sure others will find this feature useful, too. 

Thank you very much in advance, from dev to dev.
Alex

arkypita, 2024-10-22:
Did you know about LaserGRBL telegram bot?

AlexAwesome88, 2024-10-23:
Yes, but I can't find it on Telegram. Is it still active?
[image]

I just thought that adding a call to a postscript or executeable would give us freedom and independence from messenger apps. Plus it would add reliability.

Maybe you could give me a hint to the source file where I could insert the call myself. I'm a C++ Dev but I think I can get it done in C#

arkypita, 2024-10-23:
[image]

Hostaler, 2024-10-23:
What is the list of commands for lasergrbl-bot?

arkypita, 2024-10-23:
https://lasergrbl.com/telegram/

Hostaler, 2024-10-24:
Woooww thank you so much.

AlexAwesome88, 2024-10-25:
Thank you for the screenshot to the Bot. o.O I was searching on Telegram with the string "LaserGRBL" without the "_Bot" and it didn't come up although only a couple of entries appeared in the list. Telegram's search algo is not that good apparently.
Anyway, I will use the bot for now.

I'd still appreciate the option to execute a postscript or start an executeable myself just to be super flexible. Especially Devs will appreciate that feature a lot! 

Maybe you'll think about it. I'm sure it won't be a big implementation task.