arkypita/LaserGRBL #2393: No Z Up/Down control in Windows 11

ReggiePye, 2024-10-29:
**Describe the bug**
In Windows 10 the Z up/Down control is visible and works well both up and down in set steps in 3 stages.
In Windows 11 this up down control is not showing properly and has some differences i.e. the Up/Down control is replaced with a step size slider.
A small Z with a small arrow pointing down only.
When activated the Z will move down -0.1mm regardless of what step size is asked for.
The only way to move the Z up is by typing a command manually.

LaserGrbl v7.12.0.0
Grbl v1.1h
Wrapper: UsbSerial
Win 11 and Win 10

**To Reproduce**
Steps to reproduce the behaviour:
1. Run LaserGRBL in Windows 10 and set the settings to display Z Up/Down control
2. Run LaserGRBL in Windows 11 and set the settings to display Z Up/Down control
3. See the differences on the screen.
4. See error

**Expected behaviour**
Windows 11 should be able to show Z Up/Down control and be able to move in both directions and not have a fixed step of 0.1mm down only.

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]
This is the screen shot of Windows 11 screen.

arkypita, 2024-10-29:
Go to menu grbl, settings, and enable z controls

ReggiePye, 2024-10-29:
I have done that and it doesn't work. needs fixing for Windows 11

Sent with [Proton Mail](https://proton.me/mail/home) secure email.

ReggiePye, 2024-10-29:
[image]
Here are the setting for Windows 11

reinforce, 2024-11-03:
Same here on fresh install v7.12.0.
Z control enabled, but only Z and 0.1mm down button appears.

arkypita, 2024-11-04:
Could you please try with different DPI settings?
https://www.google.it/m?q=windows+11+dpi

ReggiePye, 2024-11-04:
That worked I can see the same as Windows 10 now, but the text is very small at 100%
Maybe some note should be made as some others have had the same problem,

reinforce, 2024-11-04:
In my case "Legacy icons" checkbox (In menu Grbl => Settings => Options) solve the problem.
I also tracked the issue with other releases and found that it started happening with version [v6.9.10](https://github.com/arkypita/LaserGRBL/tree/v6.9.10) version. Earlier releases works fine.

ReggiePye, 2024-11-04:
That is good to know, I will make a note of that.