arkypita/LaserGRBL #2399: Direction arrows wrong after homing

ReggiePye, 2024-11-03:
After doing a mpos homing the arrow direction buttons only work in a positive direction.
Before homing the machine will work flawlessly.

I tested my machine using  both LaserGRBL (latest version) and UGS

I just tried both softwares and found out that LaserGRBL seems to be the problem.
It all works fine in UGS but the problem happens in LaserGRBL so I zoomed out on the work area shown on the computer screen so I could see the cursor that represents the tool.
When I have homed and use the direction arrow buttons the cursor moves only in positive directions. So this shows me that it is NOT a hardware error, but a software error, because if it were the reverse the cursor would move in the pos and neg directions when asked.

 Also another fault is in the Z Up/Down control when using Windows 11.
Although the function has been selected in the setup it does not appear in the control window as it does in Windows 10.
I have reported this fault before and just told to activate it in the settings window which I have done, and it doesn't work.

ReggiePye, 2024-11-03:
I just downloaded Gsender and installed to try.
Same result as UGS it works fine. Hardware tests out okay but LaserGRBL fails when using after homing. Seems like the arrow keys for direction are not correct after the homing sequence.

arkypita, 2024-11-03:
https://lasergrbl.com/configuration/

arkypita, 2024-11-03:
Somtime it depends on how the machine firmware was build. Grbl (the firmware) can be settled up to have homing, and zero, at top right position, and work in negative space. This is common CNC behavior, and Grbl born as a firmware for CNC.

Laser engraver instead, follow cartesian plane logic. Zero in bottom left, x increase from left to right, y increase from near to far.

Axis direction can be changed in grbl configuration, but I believe it change also homing direction (not sure). Please refer to grbl configuration because it is not a LaserGRBL issue.

ReggiePye, 2024-11-03:
I have done all of that several times. I have spent a whole week trying to get it to work correctly.
As I have shown in the messages I have left about the issues.
Both UGS and Gsender work perfectly with the same settings in GRBL.
LaserGRBL works fine until I do a homing, then it will only go positive, even if I select the negative direction with the arrow buttons.

Sent with [Proton Mail](https://proton.me/mail/home) secure email.

ReggiePye, 2024-11-03:
Axis direction can be changed in grbl configuration, but I believe it change also homing direction (not sure). Please refer to grbl configuration because it is not a LaserGRBL issue.

I don't  feel you understand what is going on.
The homing affects the way that the direction buttons are working.
Positive X will go positive as it should.
Negative X will go positive as it should not.
Positive Y will go positive as it should.
Negative Y will go positive as it should not.
Before homing everything is correct. After homing is the problem.

arkypita, 2024-11-04:
And you probably didn't understand my answer.

LaserGRBL commands the movement to the microcontroller that is on the board, the behavior and the resulting movement depends on how the wiring is done, how the board is set up, how the firmware of the board is compiled.

The board runs a firmware called GRBL. It is the configuration of GRBL (and how it was compiled) that determines how the machine behaves, in which direction it homes, and the direction in which the axes move when a movement command is sent (before and after homing).

You need to focus on the GRBL configuration, LaserGRBL has nothing to do with this problem.

arkypita, 2024-11-04:
Please refer to GRBL firmware documentation.

https://github.com/gnea/grbl/wiki/Set-up-the-Homing-Cycle

https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration

https://github.com/gnea/grbl/blob/master/grbl/config.h

arkypita, 2024-11-04:
What is the position reported after homing? Top right of LaserGRBL windows.

ReggiePye, 2024-11-04:
I can send you a short screen capture movie that shows everything.
I have homed and then used the arrow direction keys to move the tool.
The screen is full so you can also see the reported positions as well.
https://github.com/user-attachments/assets/dd2d88f9-d4ad-4313-b9e1-4851c084d32b

ReggiePye, 2024-11-04:
I also flashed GRBL from LaserGRBL 1.1h original
I have followed all those links that you sent me and several more during this last week.
What I don't understand is why it all works in UGS and gSender without any problems.
[Laser K30 machine.txt](https://github.com/user-attachments/files/17616304/Laser.K30.machine.txt)

ReggiePye, 2024-11-04:
> What is the position reported after homing? Top right of LaserGRBL windows.

[image]

ReggiePye, 2024-11-05:
I redownloaded GRBL 1.1h and edited config.h line 129 and now it is working 0 0 0

ReggiePye, 2024-11-05:
Did you read the file [Laser K30 machine.txt](https://github.com/user-attachments/files/17616304/Laser.K30.machine.txt)
It shows what settings I have.

ReggiePye, 2024-11-05:
This morning I used UGS wizard to configure the settings in GRBL.
In UGS the machine works properly and does a homing and then the arrow direction keys all work as they should.

I exported the settings and carefully copied the content into a text file.
I renamed the text file to be a .NC file and loaded that into LaserGRBL and ran it.
LaserGRBL  inserted all the 34 commands into it's setting with no errors.

I then rebooted the computer and open LaserGRBL again and connected to my machine.
I didn't home to start with. I just tested the movements with the arrow buttons and they all worked fine.
I then did a homing cycle and the homing worked as expected going to the front left.

I then tried using the arrow buttons and again the movements are only positive even if I use in the negative direction.
If I keep moving in a positive direction until I reach positive space the arrow buttons will work in the negative direction.

The settings exported from LaserGRBL that are giving the problem, but only in LaserGRBL:

$0=10
$1=255
$2=2
$3=3
$4=0
$5=0
$6=0
$10=19
$11=60.000
$12=0.002
$13=0
$20=0
$21=1
$22=1
$23=3
$24=50.000
$25=1000.000
$26=250
$27=3.000
$30=1000
$31=0
$32=1
$100=80.000
$101=80.000
$102=640.000
$110=6000.000
$111=6000.000
$112=500.000
$120=1200.000
$121=1200.000

I need to use homing in LaserGRBL. The machine will work ok without homing, but I have spent a very lot of money to build a machine with Z axis that requires a working homing cycle and I need this  problem to be sorted please.

Hostaler, 2024-11-06:
The parameter $120 is not written correctly, it should say $120=120,000. This could be one of the factors, but wait to see what the experts say.

ReggiePye, 2024-11-06:
Thanks for noticing that, I will make a correction.

ReggiePye, 2024-11-07:
I have fixed the problem I think.
I was not sure which Uno board I was using, they all look the same. To make sure I edited config.h in GRBL by uncommenting the line 129.
Now when I do a homing the mpos shows 0 0 0 and the arrow buttons work properly, I then enter a G92 X0 Y0 to bring the wpos home to the same location, and everything is now working fine.

Hostaler, 2024-11-09:
> edité config.h en GRBL descomentando la línea 129.

where did you find line 129?

StuartB4, 2024-11-09:
[image]

ReggiePye, 2024-11-09:
That is correct. I uncommented that line, recompiled and reprogramed my Uno.
After that the homing worked okay. I tested it and it worked properly.

Then I restarted my controller, rehomed again and before moving any axis I typed in G10 L20 P1 X0 Y0
that positioned the wpos to be the same as the mpos home.
To make it so that I don't have to do that each time I home I entered $10=0
That stores it permanently for each homing.

closed 2024-12-01 by ReggiePye