arkypita/LaserGRBL #2408: Open Project with multiple copies

bpierangelo, 2024-11-12:
I have imported a png image file, applied parameters to suit the wanted quality in the "Import Raster Image" windows.
In the "Target Image" windows I have set my speed and laser power options,

I have appended the same png file, but this time, in the "Target Image", under "Image Size and Position" I have specified an offset in X & Y, and have done this several times to fit as many copies as possible in my ply sheet.

I have saved the project.
When I open that project, not all copies of my image are imported, and the strange thing is that if I open the file again, then random different copies of my image are imported.

For example, let's say I have 20 copies of my image, named as 1,2....20, in a array of 10 x 2.
Sometimes, it would open 1, 2 & 5, sometimes just copy 1, and sometimes  copy 1, 3, 8 or 1,3 ,5 ,6. There is not common pattern, and for sure not all instances are imported.

In older versions of Laser gbrl, I was able to open a saved project and all copies of my image would open successfully.

mic-mess, 2024-12-10:
I have a similar problem.
Saved a file with two images imported (the first one vectorized, the second one Line to Line).
When I open the file, LaserGRBL 7.14.0 opens only the first image.
If I rollback to version 7.8.0, it opens all the images.

arkypita, 2024-12-11:
Please share your project file so i can test it

mic-mess, 2024-12-11:
Here is a file with this behaviour.
This file works with LaserGRBL 7.8.0
[SegnalibroStella Angelica.zip](https://github.com/user-attachments/files/18101282/SegnalibroStella.Angelica.zip)