arkypita/LaserGRBL #2412: Limit switch during work

Jorge1000000, 2024-11-17:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Jorge1000000, 2024-11-17:
I just added a limit switch and started to use laser but the work stopped when reached X limit switch.
Is there any way to use the entire work space and bypass the limit swichs during laser work?

Hostaler, 2024-11-18:
You can send images of:
1.- general view of the machine.
2.- view along the X axis where the stroke limit can be seen.
3.- view along the Y axis where the stroke limit can be seen.
But I have to make something clear to you, if you remove the stroke limits the machine will move beyond the stroke limits and may get stuck or in the case that it goes beyond the stroke limit and touches the structure of the machine for a hundredth of a second the image you are working on is no longer valid since you lose steps.