arkypita/LaserGRBL #2422: sudden connection problem

aartnl, 2024-11-22:
I have been using laserGBRL with a laser engraver (iKier K1 Pro 1064nm 20W Fiber Laser) for months and it has always worked fine (system Windows 11). Now laserGBRL suddenly can no longer connect to the laser engraver. Lightburn does work.

Start LaserGBRL and click 'connect'.

Error message:
[image]

Screenshot LaserGRBL:
[image]

Sessionlog:

21-11-2024 21:08:52.123    Program         ------- LaserGRBL v7.12.0 [64bit] START -------
21-11-2024 21:08:52.579    OpenGL          AUTO->FBO OpenGL 4.5.0 - Build 24.20.100.6026, Intel, Intel(R) UHD Graphics 630
21-11-2024 21:08:59.098    SetStatus       Machine status [Connecting]
21-11-2024 21:08:59.102    OpenCom         Open COM3 @ 115200 baud Ctrl-X (UsbSerial)
21-11-2024 21:08:59.108    SendImmediate    Send Immediate Command [0x18]
21-11-2024 21:08:59.112    CloseCom        Close COM3 [CORE]
21-11-2024 21:08:59.115    SetStatus       Machine status [Disconnected]

Does anyone have any idea what could be wrong? Thanks in advance!

Regards,
Aart

khturnings, 2024-11-22:
Check you are using  the right COM port - I have unplugs mine a few times and then plugged it back in and windows in its wisdom had changed COM port number and so I needed to reselect in LaserGrBL

arkypita, 2024-11-22:
Try this: menu grbl, LaserGRBL settings, in first page enable both hard and soft reset on connect options.

aartnl, 2024-11-22:
Thanks, khtirnings and arkypita! Unfortunately this does not solve the problem

Newronio, 2024-11-24:
same problem here, even installing older versions.....   also my lightburn works very well

PhxPhoton, 2024-11-25:
I was using Laser GRBL without issue, and ran a windows update during a break between burns. After the update, the identical problem at the start of this thread occurred. Switched to an older PC without any windows update, and the same revision of Laser GRBL worked flawlessly. It seems like MS changed something....

Someday226, 2024-11-25:
I am having the same issue. I have tried everything I could think of but will not work, I have etsy orders to complete and I do not no what to do now. I have tried the things suggested in this thread with no joy.  Has anyone got any other suggestions for us?

JoJo-Duffy, 2024-11-25:
Any update on this -- same exact issue here -

Someday226, 2024-11-25:
Yes I have sorted it
If you go onto the program at the top you will see tools you need to download the new driver from there and then mine worked I spent 7 hours working on it but try it and let me know
Thanks

JoJo-Duffy, 2024-11-25:
Thank you...under Tools, is it install ch340 driver?

I tried that and it failed install

Someday226, 2024-11-25:
Ahhh have you had any problems before as I have used it for 3 years and no problems until now

JoJo-Duffy, 2024-11-25:
None at all..5 yr user...tried another laptop..  same issue....

Someday226, 2024-11-25:
Well I uninstalled the program first then I went to tools maybe

Newronio, 2024-11-25:
tried to solve reinstalling the new 2024 CH340 driver....   nothing
happens.. same comm issue.

Someday226, 2024-11-25:
Unfortunately i don’t know what else to try sorry wish I could help more

arkypita, 2024-11-26:
I also suggest to switch to UsbSerial2

[image]

arkypita, 2024-11-26:
Also try "run as administrator" https://www.windowscentral.com/how-run-app-administrator-windows-10

arkypita, 2024-11-26:
I also found this: [image]

https://www.arduined.eu/ch340-windows-10-driver-download/?fbclid=IwY2xjawGyp2lleHRuA2FlbQIxMQABHfVORF5tOEeVi4aGFqoF2WqzmMFtwqPTBdLXj1YoR7TI9ruIg2LfXJ2QNw_aem_S4doA5Hod8Vt6QQjEp_e5A

Newronio, 2024-11-26:
> I also found this: Screenshot_2024-11-26-13-58-45-464_com.facebook.katana.jpg
> (view on web)
>
>
> https://www.arduined.eu/ch340-windows-10-driver-download/?fbclid=IwY2xjawGyp2lleHRuA2FlbQIxMQABHfVORF5tOEeVi4aGFqoF2WqzmMFtwqPTBdLXj1YoR7TI9ruIg2LfXJ2QNw_aem_S4doA5Hod8Vt6QQjEp_e5A

nothing works.... administrator, reinstall ch340, Usbserial2.... nothing...
already I'm trying to do some things with Lightburn....... not easy....

anyway.. thanks....

StuartB4, 2024-11-26:
Maybe you need the CH341ser driver.

Newronio, 2024-11-26:
Tried also!!!

Anyone here could solve this bug?

StuartB4, 2024-11-26:
Oh ok. 👍

2024sgit, 2024-11-26:
I had to uninstall the 2024 version of the CH340 driver that came with the windows update and then installed the 2019 version using the tools tab in Lasergrbl. Just doing the install did not work. Check in device manager when connected to make sure the correct version is installed.

Someday226, 2024-11-26:
Any luck

2024sgit, 2024-11-26:
Yes. Doing the uninstall off 2024 and install of 2019 CH340 driver fixed the problem and Lasergrbl now connects to my Longer laser.

Newronio, 2024-11-26:
no luck yet....
one more time, I will uninstall grbl ( actually testing v7.11.0),   reset
all, uninstall drivers then install everything again from zero. etc etc
etc........

hope it will  work back again...

as jugglers says....    Eu acreditoooo!

Someday226, 2024-11-27:
Awesome

JoJo-Duffy, 2024-11-28:
I'm learning lightburn...i can't afford to be down any longer

alastaira, 2024-12-04:
> […](#)

Don't install the new 2024 driver... that's where the problem is! You need to rollback to a previous version, like this. 
[image]

Newronio, 2024-12-04:
I don´t know what f... happens.
magically came back to live .
everything is normal now...

try some ways...   uninstall and reinstall everything,  drivers, programs
etc
uninstall some windows updates and reinstalled later.....  nothing
happens.....

work only with lightburn for at least one week.....
and today morning....    voilà.....   its magic!!!

aartnl, 2024-12-25:
after the last update of LaserGBRL (V7.14) everything works perfectly again! Thanks!!

closed 2024-12-25 by aartnl