arkypita/LaserGRBL #2425: Welcome to LaserGRBL Discussions!

DavePut, 2024-11-26:
### Discussed in https://github.com/arkypita/LaserGRBL/discussions/1209

<div type='discussions-op-text'>

<sup>Originally posted by **arkypita** December 16, 2020</sup>
## 👋 Welcome!

If you have any doubts about LaserGRBL or are looking for help to configure and use your engraver, you can search through the opened discussions: often the answers are already there.

If you don't find an answer you can open a new one.
Remember, to get help more easily always describe:
- which engraver do you have
- what are you trying to do
- what attempts have you already made
- at what point do you stop and what message/error do you have

We’re using Discussions as a place to connect with other members of our community. We hope that you:
- Ask questions you’re wondering about.
- Share ideas.
- Engage with other community members.
- Welcome others and are open-minded, remember that this is a community we build together 💪.

</div>

DavePut, 2024-11-26:
I am new to Laser Engraving so I thought as a first test I should run the LaserGRBL power/speed test
I did not set any parameters just left the defaults, but when I ran it I aborted very quickly as F100/S700 was burning right through my 4mm plywood and I just wanted to check the colours for engraving. My Sculpfun iCube is nominally 10W and max speed 10,000 mm/min
I wonder if anyone would like to suggest a test range for power and speed before I try again.

Hostaler, 2024-11-26:
In the Lasergrbl menu you have several tests to try.

DavePut, 2024-11-27:
Appreciate that there are several tests, but I was looking for some guidance for upper and lower bounds for Engraving rather than waste a lot of wood

Hostaler, 2024-11-27:
Send your machine home, turn on the laser in the corner where the machine is now, put the wood where the laser point is, load the image into lasergrbl and move it to the corner where you have the laser but on the computer screen. This way you have everything within the tightest limits.