arkypita/LaserGRBL #2437: CH340 driver issue - worked yesterday but not today

RRCraftworks, 2024-12-02:
**Describe the bug**
Uninstalled widows and had no issue yesterday. 
Today everything works but the laser will not come on. 

**To Reproduce**
set a jox and will do everything but work. 

**Expected behavior**
I engraved for hours yesterday after spending 3 hrs figuring out a work around. today I can connect but now it will not light up.

**Hardware and software configuration (please complete the following information):**
 - Windows version 11 uninstalled update. tried reinstalling update and using ch340 patch installed..
 - Engraver brand and model Fox Alien 20w
 - Grbl version latest.... tried updating that too.
 - LaserGRBL version

**Additional context**

Please fix this issue..... We have so many craft fairs coming up...

alastaira, 2024-12-04:
You can rollback to a previous version of the CH340 driver. [image]

kekosinclair, 2024-12-04:
Thanks for the help. My Laser is like this too. I tried to revert but the return button is disabled. I really need to resolve this because I have a lot of projects in progress.

alastaira, 2024-12-04:
In that case you can click the Disable Device button instead, and check the button that says to also remove the driver. Then, install the (older) CH340 from the LaserGRBL menu.