arkypita/LaserGRBL #2463: LaserGRbl crashes if not used during a while

Mixailovitch, 2024-12-17:
Hello,
Firstly thank yo uvery much for sharing this program. It's fine.

I have just a problem with last version as you can see in the error report.
I have my laser in a separate room  and I do not close LaserGrbl when I leaves the room after finishing the job without problem.
When, I come back later it has crashed meanwhile with following error report :

LaserGrbl v7.14.0.0
Grbl v1.1h
Wrapper: UsbSerial
Win 10, Home, 10.0.19045.0 (7)
CLR: C:\Windows\Microsoft.NET\Framework64\v4.0.30319\mscorlib.dll 4.8.9282.0 built by: NET481REL1LAST_C

TypeOf exception  [System.ObjectDisposedException]
Exception message [Le handle sécurisé a été fermé]
Exception source  [mscorlib], thread [Main Thread]
Exception method  [Void DangerousAddRef(Boolean ByRef)]

   ----------- stack trace -----------
   à System.Runtime.InteropServices.SafeHandle.DangerousAddRef(Boolean& success)
   à System.StubHelpers.StubHelpers.SafeHandleAddRef(SafeHandle pHandle, Boolean& success)
   à Microsoft.Win32.UnsafeNativeMethods.GetOverlappedResult(SafeFileHandle hFile, NativeOverlapped* lpOverlapped, Int32& lpNumberOfBytesTransferred, Boolean bWait)
   à System.IO.Ports.SerialStream.EventLoopRunner.WaitForCommEvent()
   à System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   à System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   à System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
   à System.Threading.ThreadHelper.ThreadStart()

Thank you

Nicolas

Mixailovitch, 2024-12-17:
It happens each time (always)

arkypita, 2025-03-01:
thanks to all

This DangerousAddRef error is inside a windows component System.IO.Ports.SerialStream that is not under my control, and I cannot fix it directly because it is not LaserGRBL code.

I suggest all of you to update the latest version of [Microsoft .NET framework](https://www.google.com/search?client=opera&q=Microsoft+.NET+framework&sourceid=opera&ie=UTF-8&oe=UTF-8) and if not work, to change in LaserGRBL settings from UsbSerial to RJCPSerial

[image]