arkypita/LaserGRBL #2470: Your latest update created problems

BoxJoint24, 2024-12-21:
**Describe the bug**
I have been using Laser GRBL for several years and have had great results mainly marking on wood surfaces. After installing the latest firmware update, things went crazy. A project will start correctly, but at some point it starts producing gibberish like images and will shift parts of a line of text to the right. 
Any suggestions?
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware and software configuration (please complete the following information):**
 - Windows version     11
 - Engraver brand and model    Not sure of brand or model  3.5w
 - Grbl version (i.e. 0.9j, 1.1f)
 - LaserGRBL version                      V7.14.0

**Additional context**
Add any other context about the problem here.

BoxJoint24, 2024-12-22:
Here is another example of the shift. I would like to know how to fix this problem.
[image]

arkypita, 2024-12-22:
You can easly rollback to a previous version: https://github.com/arkypita/LaserGRBL/releases

BoxJoint24, 2024-12-23:
Thank you for the reply and the suggestion. I suppose I'll have to remove the latest version first. Before I do that , I need to determine that the laser itself is not failing.

closed 2024-12-23 by BoxJoint24

arkypita, 2024-12-23:
Remove the opd one is not really necessary, but if you want you can do it. Im ny opinion it is not a LaserGRBL issue, but it is your hardware fault. I say that, because none of the changes I did to LaserGRBL can cause your issue. However, feel free to test with other versions and let me know.

BoxJoint24, 2024-12-23:
Thanks for your help. I am having trouble finding the download for v13. Can you send me the link please?Mike

arkypita, 2024-12-23:
https://github.com/arkypita/LaserGRBL/releases/download/v7.13.0/install.exe

BoxJoint24, 2024-12-23:
Your help is appreciated.I think I may have changed some settings in the GRBL config. Now I am getting mirror images.Can you send me an image of the default config settings so I can get back to a known starting point?I am sorry to be of so much trouble.Thank You so much,Mike

 
 

https://github.com/arkypita/LaserGRBL/releases/download/v7.13.0/install.exe

BoxJoint24, 2024-12-27:
I installed v7.13.0 and it appears to have cleared up the problem. I will let you know if anything changes.Thanks so much for your help.Mike Sweeney,A loyal subscriber

 
 

https://github.com/arkypita/LaserGRBL/releases/download/v7.13.0/install.exe