arkypita/LaserGRBL #2472: error report

mana4215, 2024-12-21:
I can complete the project but this error report comes up after project is complete.  I don't know what it means.

**Describe the bug**
A clear and concise description of what the bug is.

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Go to grbl project
2. run engraving (burn to wood)project
3. this error comes up after project is complete.

5. Click on '....'
6. Scroll down to '....'
7. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware and software configuration (please complete the following information):**
 - Windows version
 - Engraver brand and model
 - Grbl version (i.e. 0.9j, 1.1f)
 - LaserGRBL version

**Additional context**
Add any other context about the problem here.

flexy-grbl1987, 2024-12-23:
LaserGrbl v7.14.0.0
Grbl v1.1h
Wrapper: UsbSerial
Win 11, Home, 10.0.22631.0 (7)
CLR: C:\Windows\Microsoft.NET\Framework64\v4.0.30319\mscorlib.dll 4.8.9282.0 built by: NET481REL1LAST_C

TypeOf exception  [System.ObjectDisposedException]
Exception message [Se ha cerrado el controlador seguro]
Exception source  [mscorlib], thread [Main Thread]
Exception method  [Void DangerousAddRef(Boolean ByRef)]

   ----------- stack trace -----------
   en System.Runtime.InteropServices.SafeHandle.DangerousAddRef(Boolean& success)
   en System.StubHelpers.StubHelpers.SafeHandleAddRef(SafeHandle pHandle, Boolean& success)
   en Microsoft.Win32.UnsafeNativeMethods.GetOverlappedResult(SafeFileHandle hFile, NativeOverlapped* lpOverlapped, Int32& lpNumberOfBytesTransferred, Boolean bWait)
   en System.IO.Ports.SerialStream.EventLoopRunner.WaitForCommEvent()
   en System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   en System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   en System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
   en System.Threading.ThreadHelper.ThreadStart()

arkypita, 2025-03-01:
thanks to all

This DangerousAddRef error is inside a windows component System.IO.Ports.SerialStream that is not under my control, and I cannot fix it directly because it is not LaserGRBL code.

I suggest all of you to update the latest version of [Microsoft .NET framework](https://www.google.com/search?client=opera&q=Microsoft+.NET+framework&sourceid=opera&ie=UTF-8&oe=UTF-8) and if not work, to change in LaserGRBL settings from UsbSerial to RJCPSerial

[image]

flexy-grbl1987, 2025-03-01:
Thank you

Obtener Outlook para Android<https://aka.ms/AAb9ysg>

image.png (view on web)<https://github.com/user-attachments/assets/33a6b0ec-9cdd-46f6-a003-9780cfcd5c55>

image.png (view on web)<https://github.com/user-attachments/assets/33a6b0ec-9cdd-46f6-a003-9780cfcd5c55>