arkypita/LaserGRBL #2509: Message Ooops Something went wrong

Davey530311, 2025-01-29:
LaserGrbl v7.14.0.0
Grbl v1.1f
Wrapper: UsbSerial
Win 10, Home, 10.0.19045.0 (7)
CLR: C:\Windows\Microsoft.NET\Framework64\v4.0.30319\mscorlib.dll 4.8.9290.0 built by: NET481REL1LAST_C

TypeOf exception  [System.ObjectDisposedException]
Exception message [Safe handle has been closed]
Exception source  [mscorlib], thread [Main Thread]
Exception method  [Void DangerousAddRef(Boolean ByRef)]

   ----------- stack trace -----------
   at System.Runtime.InteropServices.SafeHandle.DangerousAddRef(Boolean& success)
   at System.StubHelpers.StubHelpers.SafeHandleAddRef(SafeHandle pHandle, Boolean& success)
   at Microsoft.Win32.UnsafeNativeMethods.GetOverlappedResult(SafeFileHandle hFile, NativeOverlapped* lpOverlapped, Int32& lpNumberOfBytesTransferred, Boolean bWait)
   at System.IO.Ports.SerialStream.EventLoopRunner.WaitForCommEvent()
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Threading.ThreadHelper.ThreadStart()

marialuzfi, 2025-02-06:
LaserGrbl v7.14.0.0
Grbl v1.1h
Wrapper: UsbSerial
Win 10, Home, 10.0.19045.0 (7)
CLR: C:\Windows\Microsoft.NET\Framework64\v4.0.30319\mscorlib.dll 4.8.9290.0 built by: NET481REL1LAST_C

TypeOf exception  [System.ObjectDisposedException]
Exception message [Se ha cerrado el controlador seguro]
Exception source  [mscorlib], thread [Main Thread]
Exception method  [Void DangerousAddRef(Boolean ByRef)]

   ----------- stack trace -----------
   en System.Runtime.InteropServices.SafeHandle.DangerousAddRef(Boolean& success)
   en System.StubHelpers.StubHelpers.SafeHandleAddRef(SafeHandle pHandle, Boolean& success)
   en Microsoft.Win32.UnsafeNativeMethods.GetOverlappedResult(SafeFileHandle hFile, NativeOverlapped* lpOverlapped, Int32& lpNumberOfBytesTransferred, Boolean bWait)
   en System.IO.Ports.SerialStream.EventLoopRunner.WaitForCommEvent()
   en System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   en System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   en System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
   en System.Threading.ThreadHelper.ThreadStart()

brodao2, 2025-02-07:
The same error
LaserGrbl v7.14.0.0
Grbl v1.5D
Wrapper: UsbSerial
Win 11, Home, 10.0.26100.0 (7)
CLR: C:\Windows\Microsoft.NET\Framework64\v4.0.30319\mscorlib.dll 4.8.9290.0 built by: NET481REL1LAST_C

TypeOf exception  [System.ObjectDisposedException]
Exception message [SafeHandle foi fechado]
Exception source  [mscorlib], thread [Main Thread]
Exception method  [Void DangerousAddRef(Boolean ByRef)]

   ----------- stack trace -----------
   em System.Runtime.InteropServices.SafeHandle.DangerousAddRef(Boolean& success)
   em System.StubHelpers.StubHelpers.SafeHandleAddRef(SafeHandle pHandle, Boolean& success)
   em Microsoft.Win32.UnsafeNativeMethods.GetOverlappedResult(SafeFileHandle hFile, NativeOverlapped* lpOverlapped, Int32& lpNumberOfBytesTransferred, Boolean bWait)
   em System.IO.Ports.SerialStream.EventLoopRunner.WaitForCommEvent()
   em System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   em System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   em System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
   em System.Threading.ThreadHelper.ThreadStart()

Lausolsi, 2025-02-14:
LaserGrbl v7.14.0.0
Grbl v1.1h
Wrapper: UsbSerial
Win 10, Home, 10.0.19045.0 (7)
CLR: C:\Windows\Microsoft.NET\Framework64\v4.0.30319\mscorlib.dll 4.8.9290.0 built by: NET481REL1LAST_C

TypeOf exception  [System.ObjectDisposedException]
Exception message [Se ha cerrado el controlador seguro]
Exception source  [mscorlib], thread [Main Thread]
Exception method  [Void DangerousAddRef(Boolean ByRef)]

   ----------- stack trace -----------
   en System.Runtime.InteropServices.SafeHandle.DangerousAddRef(Boolean& success)
   en System.StubHelpers.StubHelpers.SafeHandleAddRef(SafeHandle pHandle, Boolean& success)
   en Microsoft.Win32.UnsafeNativeMethods.GetOverlappedResult(SafeFileHandle hFile, NativeOverlapped* lpOverlapped, Int32& lpNumberOfBytesTransferred, Boolean bWait)
   en System.IO.Ports.SerialStream.EventLoopRunner.WaitForCommEvent()
   en System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   en System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   en System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
   en System.Threading.ThreadHelper.ThreadStart()

arkypita, 2025-03-01:
thanks to all

This DangerousAddRef error is inside a windows component System.IO.Ports.SerialStream that is not under my control, and I cannot fix it directly because it is not LaserGRBL code.

I suggest all of you to update the latest version of [Microsoft .NET framework](https://www.google.com/search?client=opera&q=Microsoft+.NET+framework&sourceid=opera&ie=UTF-8&oe=UTF-8) and if not work, to change in LaserGRBL settings from UsbSerial to RJCPSerial

[image]

fernandonaim, 2025-03-25:
Has anyone managed to solve it? I have the same problem and nothing

LaserGrbl v7.14.1.0
Grbl v1.5D
Wrapper: RJCPSerial
Win 11, Professional, 10.0.26100.0 (7)
CLR: C:\Windows\Microsoft.NET\Framework64\v4.0.30319\mscorlib.dll 4.8.9290.0 built by: NET481REL1LAST_C

TypeOf exception  [System.ArgumentOutOfRangeException]
Exception message [InvalidArgument=Value '0' não é um valor válido para 'index'.
Nome do parâmetro: index]
Exception source  [System.Windows.Forms], thread [Main Thread]
Exception method  [System.Object get_Item(Int32)]

   ----------- stack trace -----------
   em System.Windows.Forms.ComboBox.ObjectCollection.get_Item(Int32 index)
   em System.Windows.Forms.ComboBox.get_Text()
   em System.Windows.Forms.ComboBox.WmReflectCommand(Message& m)
   em System.Windows.Forms.ComboBox.WndProc(Message& m)
   em LaserGRBL.UserControls.FlatComboBox.WndProc(Message& m) na E:\LaserGRBL\LaserGRBL\UserControls\FlatComboBox.cs:linha 14
   em System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)