arkypita/LaserGRBL #2527: idea for a better generators

maot1985, 2025-03-04:
The generators are a great idea, but not practical for carrying out initial tests on materials.

Explained using the example of “intensity vs. speed”. The test file created simply runs through the engraving from left to right, line by line. If there is a problem, for example because the intensity is too high at a certain point, you have to stop and have not yet tested other settings with, for example, more speed and less intensity.

I wrote my own routine in Excel using VBA that goes through the individual combination fields diagonally.

Example: My created test table has 3 columns with column1=S100, column2=S200, column3=S300 and 3 rows with row1=F2000, row2=F1500, row 3=F1000. The origin of the table is in the left bottom corner.

The order of the individual test fields for me is:
S100/F2000
S200/F2000
S100/F1500
S300/F2000
S200/F1500
S100/F1000
S300/F1500
S200/F1000
S300/F1000

I attached my VBA-code if someone wants to optimize the generators:

'######################################################################################################

Sub Gravur_Muster_Automatik(Fmax As Long, Fmin As Long, Fdelta As Long, Smax As Long, Smin As Long, Sdelta As Long, dist As Double, L As Double, p As Double, xstart As Double, ystart As Double, Datei As String)

    '######### Deklarierung Arrays

    Dim werte_F As Collection
    Set werte_F = New Collection
    Dim werte_s As Collection
    Set werte_s = New Collection
    Dim werte_posx As Collection
    Set werte_posx = New Collection
    Dim werte_posy As Collection
    Set werte_posy = New Collection

    '######### Feldgröße S und F ermitteln

    Dim Fn As Integer
    Dim Sn As Integer
    Fn = (Fmax - Fmin) / Fdelta + 1
    Sn = (Smax - Smin) / Sdelta + 1

    '######### Reihenfolge festlegen > wertex() und wertey()

    Call DiagonaleZeilenSpalten(Fn, Sn)
    
    '######### X und Y Werte erzeugen abhängig von Feldgrößen, Abstand und Länge
    
    Dim bS As Integer
    Dim bF As Integer
    
    For bS = 1 To Sn
        werte_s.Add Smin + (bS - 1) * Sdelta
        werte_posx.Add dist * (bS - 1)
    Next bS
    
    For bF = 1 To Fn
        werte_F.Add Fmin + (bF - 1) * Fdelta
        werte_posy.Add (Fn - 1) * dist - dist * (bF - 1)
    Next bF
     
    '######### Beginn Schleife
    
    Dim n As Integer
    Dim nmax As Integer
    nmax = wertex.Count
    
    Dim F As Long
    Dim S As Long
    Dim x As Double
    Dim y As Double
    
    Print #Dateinummer, "(Startpunkt X=" & xstart & " / Y=" & ystart & ")"
    Print #Dateinummer, "(--------------------------)"
    Print #Dateinummer, "G0 X" & xstart & " Y" & ystart & " F2000"
        
    For n = 1 To nmax
        F = werte_F(wertey(n))
        S = werte_s(wertex(n))
        x = werte_posx(wertex(n)) + xstart
        y = werte_posy(wertey(n)) + ystart
       
        Call Gravur_Quadrat(x, y, L, S, F, p)

    Next n
    
    Print #Dateinummer, "(Endpunkt X=" & xstart & " / Y=" & ystart & ")"
    Print #Dateinummer, "(--------------------------)"
    Print #Dateinummer, "G0 X" & xstart & " Y" & ystart & " F2000"
    
End Sub

'######################################################################################################

Sub Gravur_Quadrat(x1 As Double, y1 As Double, L As Double, S As Long, F As Long, p As Double)
    
    Dim i As Integer
    Dim yvar As Double
    
    Dim x2 As Double
    Dim y2 As Double
    x2 = x1 + L
    y2 = y1 + L
    
    yvar = y1 + p
    
    Print #Dateinummer, "(Viereck mit S=" & S & " / F=" & F & ")"
    Print #Dateinummer, "(--------------------------)"
    Print #Dateinummer, "G0 X" & x1 & " Y" & y1 & " F" & F
    Print #Dateinummer, "M4 S0"
    Print #Dateinummer, "(For)"
    For i = 1 To 100
        Print #Dateinummer, "(>>>)"
        Print #Dateinummer, "G1 X" & x2 & " S" & S                  'Fahrt nach Rechts
        Print #Dateinummer, "S0"                                    'Laser aus
        Print #Dateinummer, "G0 X" & x2 & " Y" & yvar & " S0"       'Fahrt um p nach oben
        yvar = yvar + p                                             'p erhöhen
        Print #Dateinummer, "G1 X" & x1 & " S" & S                  'Fahrt nach Links
        Print #Dateinummer, "S0"                                    'Laser aus
        Print #Dateinummer, "G0 X" & x1 & " Y" & yvar & " S0"       'Fahrt um p nach oben
        yvar = yvar + p                                             'p erhöhen
        Print #Dateinummer, "(>>>)"
        If yvar > y2 Then Exit For
    Next i
    Print #Dateinummer, "(End For)"
    Print #Dateinummer, "M5"
    Print #Dateinummer, ""
    
End Sub

'######################################################################################################

Sub DiagonaleZeilenSpalten(r As Integer, c As Integer)

    Dim row As Integer, col As Integer
    Dim maxRow As Integer, maxCol As Integer
    Dim startRow As Integer, startCol As Integer
    
    Set wertex = New Collection
    Set wertey = New Collection
    
    maxRow = r ' Anzahl der Zeilen
    maxCol = c ' Anzahl der Spalten
    
    ' Erste Phase: Starte in der ersten Spalte (von unten nach oben)
    For startRow = maxRow To 1 Step -1
        row = startRow
        col = 1
        Dim values_x As Collection
        Set values_x = New Collection
        Dim values_y As Collection
        Set values_y = New Collection

        ' Werte in umgekehrter Reihenfolge speichern
        Do While row <= maxRow And col <= maxCol
            values_x.Add col
            values_y.Add row
            row = row + 1
            col = col + 1
        Loop

        ' Werte in umgekehrter Reihenfolge ausgeben
        Dim i As Integer
        For i = values_x.Count To 1 Step -1
            wertex.Add values_x(i)
            wertey.Add values_y(i)
        Next i
        
    Next startRow
    

    ' Zweite Phase: Starte in der obersten Zeile (von links nach rechts, außer der ersten Spalte)
    For startCol = 2 To maxCol
        row = 1
        col = startCol
        Set values_x = New Collection
        Set values_y = New Collection
        
        ' Werte in umgekehrter Reihenfolge speichern
        Do While row <= maxRow And col <= maxCol
            values_x.Add col
            values_y.Add row
            row = row + 1
            col = col + 1
        Loop

        ' Werte in umgekehrter Reihenfolge ausgeben
        For i = values_x.Count To 1 Step -1
            wertex.Add values_x(i)
            wertey.Add values_y(i)
        Next i
    
        
    Next startCol
    
End Sub

maot1985, 2025-03-04:
[Laser-Muster.zip](https://github.com/user-attachments/files/19067119/Laser-Muster.zip)

ristraus, 2025-03-05:
Very useful, but if you allow me a suggestion, what about add the values description, like LaserGRBL adds when you generate a test file ?
Danke !!!!

maot1985, 2025-03-05:
I wanted to add a value description, but I have not found yet a simple solution how to generate gcode from text in excel VBA.
If I find a solution I add it here ;-)

At the moment I use the visualisation in the Excel-file.

I copy the entire range and paste it into Paint.net.

The so generated image I use for LaserGRBL to create the value description :-D

ristraus, 2025-03-06:
I have tried ChatGPT plus  DeepSeek for ideas on how to do it, but not really nice answers. :-(

maot1985, 2025-03-06:
[Laser-Muster v2.zip](https://github.com/user-attachments/files/19103353/Laser-Muster.v2.zip)

I also tried ChatGPT, but got no solution for excel VBA only. 
I created some characters manually and added a routine to add the values to the tables.
I'm not totally happy with it, but it will work :-)

ristraus, 2025-03-06:
"es ist nicht immer einfach!!!" ;-)
I like your solution, save some time rather need to always go back to the spreadsheet.
Wondering if the Parameter table is where you where you write down the best result obtained for a given material ? Like the material database available in LasergRBL.

ristraus, 2025-03-18:
Good Job !!!!! 
Wondering why you removed the Test-Muster spreadsheet ?
It would be nice to have it available in other languages (Although it is super easy to understand it). I was still preparing a translation table.
Why you dont make it available in the LaserGRBL Facebook group ?

Danke ;-)

[Translations.xlsx](https://github.com/user-attachments/files/19320545/Translations.xlsx)