arkypita/LaserGRBL #2550: Ts55 Not working

SteveD4590, 2025-04-02:
Hi Team, my TS55 Laser engraver will not respond to both of my computers. All software including Lightburn cannot work for me. Com ports and Baude are correct. The laser has lights on and fan motors are running. checked all cables and the seem fine albeit they are only a month old and up until yesterday everything was working fine. Note if I tour the power on using the software's the machine seems to connect fine. It just won't operate any functions.

Bleach665, 2025-06-04:
If you're talking about TwoTrees TTS-55, it works fine for me via USB. I just assembled it and am testing it)

SteveD4590, 2025-06-04:
Thanks pal. All good now it was a dead mother board.
Steve