arkypita/LaserGRBL #2555: Install bug

Pietste1, 2025-04-12:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware and software configuration (please complete the following information):**
 - Windows version
 - Engraver brand and model
 - Grbl version (i.e. 0.9j, 1.1f)
 - LaserGRBL version

**Additional context**
Add any other context about the problem here.

scott7792, 2025-04-23:
LaserGrbl v7.14.1.0
 v
Wrapper: UsbSerial
Win 11, Professional, 10.0.26120.0 (7)
CLR: C:\Windows\Microsoft.NET\Framework64\v4.0.30319\mscorlib.dll 4.8.9032.0 built by: NET481REL1

TypeOf exception  [System.Reflection.TargetInvocationException]
Exception message [Exception has been thrown by the target of an invocation.]
Exception source  [mscorlib], thread []
Exception method  [Void SerializationInvoke(System.IRuntimeMethodInfo, System.Object, System.Runtime.Serialization.SerializationInfo, System.Runtime.Serialization.StreamingContext ByRef)]

   ----------- stack trace -----------
   at System.RuntimeMethodHandle.SerializationInvoke(IRuntimeMethodInfo method, Object target, SerializationInfo info, StreamingContext& context)
   at System.Runtime.Serialization.ObjectManager.CompleteISerializableObject(Object obj, SerializationInfo info, StreamingContext context)
   at System.Runtime.Serialization.ObjectManager.FixupSpecialObject(ObjectHolder holder)
   at System.Runtime.Serialization.ObjectManager.DoFixups()
   at System.Runtime.Serialization.Formatters.Binary.ObjectReader.Deserialize(HeaderHandler handler, __BinaryParser serParser, Boolean fCheck, Boolean isCrossAppDomain, IMethodCallMessage methodCallMessage)
   at System.Runtime.Serialization.Formatters.Binary.BinaryFormatter.Deserialize(Stream serializationStream, HeaderHandler handler, Boolean fCheck, Boolean isCrossAppDomain, IMethodCallMessage methodCallMessage)
   at System.Resources.ResourceReader.DeserializeObject(Int32 typeIndex)
   at System.Resources.ResourceReader.LoadObjectV2(Int32 pos, ResourceTypeCode& typeCode)
   at System.Resources.ResourceReader.ResourceEnumerator.get_Entry()
   at System.Resources.ResourceReader.ResourceEnumerator.get_Current()
   at System.ComponentModel.ComponentResourceManager.FillResources(CultureInfo culture, ResourceSet& resourceSet)
   at System.ComponentModel.ComponentResourceManager.FillResources(CultureInfo culture, ResourceSet& resourceSet)
   at System.ComponentModel.ComponentResourceManager.ApplyResources(Object value, String objectName, CultureInfo culture)
   at LaserGRBL.ConnectLogForm.InitializeComponent() in E:\LaserGRBL\LaserGRBL\ConnectLogForm.Designer.cs:line 100
   at LaserGRBL.MainForm.InitializeComponent() in E:\LaserGRBL\LaserGRBL\MainForm.Designer.cs:line 183
   at LaserGRBL.MainForm..ctor() in E:\LaserGRBL\LaserGRBL\MainForm.cs:line 41
   at LaserGRBL.MainForm..ctor(String[] args) in E:\LaserGRBL\LaserGRBL\MainForm.cs:line 131
   at LaserGRBL.Program.Main(String[] args) in E:\LaserGRBL\LaserGRBL\Program.cs:line 60

Inner exception data
TypeOf exception  [System.TypeInitializationException]
Exception message [The type initializer for 'System.MarvinHash' threw an exception.]
Exception source  [System.Xml], thread []
Exception method  [Void .ctor()]

   ----------- stack trace -----------
   at System.Xml.NameTable..ctor()
   at System.Windows.Forms.Layout.TableLayoutSettingsTypeConverter.ConvertFrom(ITypeDescriptorContext context, CultureInfo culture, Object value)
   at System.Windows.Forms.TableLayoutSettings..ctor(SerializationInfo serializationInfo, StreamingContext context)

Inner exception data
TypeOf exception  [System.ApplicationException]
Exception message [Invalid XML in file 'C:\Windows\Microsoft.NET\Framework64\v4.0.30319\config\machine.config' near element ''.]
Exception source  [mscorlib], thread []
Exception method  [System.ConfigNode Parse(System.String, System.String, Boolean)]

   ----------- stack trace -----------
   at System.ConfigTreeParser.Parse(String fileName, String configPath, Boolean skipSecurityStuff)
   at System.Security.Cryptography.CryptoConfig.OpenCryptoConfig()
   at System.Security.Cryptography.CryptoConfig.InitializeConfigInfo()
   at System.Security.Cryptography.CryptoConfig.CreateFromName(String name, Object[] args)
   at System.Security.Cryptography.RandomNumberGenerator.Create()
   at System.MarvinHash.GenerateSeed()
   at System.MarvinHash..cctor()

Inner exception data
TypeOf exception  [System.Runtime.InteropServices.COMException]
Exception message [Error HRESULT E_FAIL has been returned from a call to a COM component.]
Exception source  [mscorlib], thread []
Exception method  [Void RunParser(System.String)]

   ----------- stack trace -----------
   at System.BaseConfigHandler.RunParser(String fileName)
   at System.ConfigTreeParser.Parse(String fileName, String configPath, Boolean skipSecurityStuff)

Dobets1, 2025-05-10:
LaserGrbl v7.14.0.0
Grbl v1.3a
Wrapper: UsbSerial
Win 11, Professional, 10.0.26100.0 (7)
CLR: C:\Windows\Microsoft.NET\Framework64\v4.0.30319\mscorlib.dll 4.8.9310.0 built by: NET481REL1LAST_C

TypeOf exception  [System.IO.IOException]
Exception message [Unzulässige Funktion.
]
Exception source  [System], thread [Main Thread]
Exception method  [Void WinIOError(Int32, System.String)]

   ----------- stack trace -----------
   bei System.IO.Ports.InternalResources.WinIOError(Int32 errorCode, String str)
   bei System.IO.Ports.SerialStream.DiscardInBuffer()
   bei System.IO.Ports.SerialStream.Dispose(Boolean disposing)
   bei System.IO.Ports.SerialStream.Finalize()