arkypita/LaserGRBL #2623: Laser does not cut due to G1Fxx command

andrewerrington, 2025-07-30:
**Describe the bug**
A clear and concise description of what the bug is.
Lasergrbl produces erroneous G1Fxx code, which leads to zero laser power when following a path.

**To Reproduce**
Steps to reproduce the behavior:
Not sure how to reproduce. I have opened two similar SVG files to send to the laser. One produces erroneous G-code, the other does not. A simple modification of the erroneous G-code allows it to work.

**Expected behavior**
A clear and concise description of what you expected to happen.
I expect to be able to load an SVG file and have the laser cut or engrave the paths therein.

**Screenshots**
If applicable, add screenshots to help explain your problem.
n/a

**Hardware and software configuration (please complete the following information):**
 - Linux Mint (recent) running LaserGRBL under Wine
 - AlgoLaser 5W DIY
 - Grbl 1.1f
 - LaserGRBL version 7.14.1

**Additional context**
Add any other context about the problem here.

Open an SVG file in LaserGRBL. Observe that laser moves but does not cut. Open a similar SVG file. Observe it cuts correctly. Export G-code for both. Examine the differences. Apart from the X-Y coordinates being different, because the shape is different, there is only one difference in the G-Code.

Basically, the beginning of the non-working file looks like this:
G90
M3 S0
S0
G0X129.85Y157.706
S500
G1F1800
X121.449Y160.01
X121.08Y160.115
.
.
.

The laser follows the path, but the laser does not turn on, despite the presence of S500 in the file.

The only difference between this and the working file is the line G1F1800. In the working file there is an XY command between G1 and F1800. Therefore, editing the non-working file and moving the first coordinate line into this line causes the file to work:
G90
M3 S0
S0
G0X129.85Y157.706
S500
G1X121.449Y160.01F1800
X121.08Y160.115
.
.
.

I deduced this from looking at the beginning of a file that worked, and a file that didn't. What caused the problem, I don't know.

I don't have access to the laser at all times, but I can get information or run tests if required.

andrewerrington, 2025-07-31:
Thanks for your reply. I don't have a "good" SVG, since they are all supposed to be good. Attached is the "broken" SVG file.

This file is part of another, larger, SVG file. The larger file is too big to cut in one operation, so individual parts are taken out and cut separately. In this case, the actions are:

- Open original SVG in Inkscape
- Ungroup all objects
- Select desired object and Copy
- Create new SVG document in Inkscape
- Paste previously copied object into new document
- Save new SVG document to be opened later in LaserGRBL
- Close original file without saving

I can't send the original file as it is a paid-for product.

The next steps are to start LaserGRBL, open SVG file, set job parameters (speed, power) and click Create. The shape from the Inkscape file correctly appears on screen. When the job is started the laser follows the path of the shape, but the laser does not fire. Upon further investigation, as stated, when looking at the G-code generated for the job we see G1F1800, whereas I believe it should be G1XnnYnnF1800 (which I gleaned by looking at a job which did cut successfully).

[image]

andrewerrington, 2025-07-31:
Hmm. I hope you can extract that SVG file from the comment. If not, I'll try something different.

mikefromgrr, 2025-08-01:
Probably won't make a difference, but when you cut-n-paste the "part" into the new Inkscape document, can you then use Inkscape's Document Properties dialog to `Resize the content` and export the document so it looks like (B) instead of (A). 

[image]

--

That said, neither of the GCODE files (from your sample or the "resized" sample, screenshot below) emitted on my workstation (not connected to an engraver) had the odd standalone `G1F1800`. Note basically `G1` turns on the laser and `F1800` is the feed rate. Agree with your conclusion that the X and Y coordinate would logically need to be in the command to engrave and move (but not saying if that is reality or not). Also, `G0` is move with the laser off.

[image]

--

Basically, I've got nothing.

andrewerrington, 2025-08-01:
I've got nothing is actually a good datapoint, thanks. It means I need to look elsewhere.

I'll set up the same applications on my own machine (the problematical machine belongs to someone else) and try exactly the same process.

Goal: Open any SVG and cut, without worrying that it might not. I realise that that should be normal operation, but I seem to have an outlier here.

mikefromgrr, 2025-08-04:
There's an optimization in the GCODE generator (as well as lots of GCODE generators, _supposedly_) that only emit X and Y coordinates when the coordinates change. I don't think the raw sample you sent results in `G1F1000` (on my machine anyway) based on what I'm seeing.

_GCODE Compression/Optimization Here:_
https://github.com/arkypita/LaserGRBL/blob/master/LaserGRBL/SvgConverter/gcodeRelated.cs#L514-L515

Regardless, I can "fake your broken file" by manually altering the sample SVG you sent **(A)** into **(B)**, where LaserGRBML emits the `G1F1000` **(C)** because the X/Y coordinate does not change between the 1st and 2nd coordinate.

[image]

From all the AI's I asked, `G1F1000` supposedly is a valid operation. I haven't tried it on my laser engraver, but tend to think that your engraver's firmware is stumbling on it...vs. a problem with LaserGRBL. I'm open to correction by any GCODE experts though!

Right now, there's no setting/pref to toggle the GCODE compression ON/OFF ([hardcoded here to "on"](https://github.com/arkypita/LaserGRBL/blob/master/LaserGRBL/SvgConverter/gcodeRelated.cs#L50)), but seemingly would be easy to add that would theoretically emit larger non-compressed GCODE to cure your problem. You could alwarys fork the repo, change that `true` to `false` and give it a shot.

I'm hoping _arkypita_ will pull in my [SVG improvement](https://github.com/arkypita/LaserGRBL/pull/2622) before I tackle other things. If I were to add a toggle, I'd leverage a [setting similar to my _Vector.ReplaceClones_ in my PR here](https://github.com/arkypita/LaserGRBL/pull/2622/commits/0a42ba11c660c9e4d16d0f5aa2dbf22587af890e).